jotego/jtbin #1: Tiger Road US Ver crashes & pixels flickering

Aquas77, 2019-11-28:
- Seemingly random crashes (MiSTer hud still accessable) only found in US Version as well as US, Romstar version.

Video examples: 
https://t.co/dZ1Sj9haGe
https://t.co/urSFJjq3w6

- Pixels are flickering in various places. One can be seen on the title screen on the bottom of the right dragon's tail.

Example on this boss's sprite in this clip: https://clips.twitch.tv/DistinctAntediluvianSangTooSpicy
And on the character's weapon in this boss room: https://clips.twitch.tv/EnthusiasticFrozenJackalPanicBasket
And on the right character's bottom left robe, and the 7 and 4 of the timer countdown: https://clips.twitch.tv/KnottyBoxyLyrebirdDuDudu

jotego, 2019-11-28:
Thank you so much for the detailed report and apologies for the bad experience.
The first release had a bug in the SDRAM controller which caused the CPU to randomly hangup. I thought I had fixed that. Could you confirm which version of the core were you using?

Aquas77, 2019-11-28:
Yes, this is the SDRAM fixed version 11/21. I'm on 64mb of sdram as well.

jotego, 2019-11-28:
Ah... Yes, I'm having problems with 64MB modules and Winbond 32MB modules. I don't have those modules myself to test them and they seem to require a slightly different delay setting in the SDRAM controller.

Anyway, before blaming it on to the memory module; let me keep the core running on MiSTer for a full day and see if it hangs up on mine. I already ran that endurance test once, but let me double check.

jotego, 2019-11-29:
Confirmed that ROMStar version hangs up randomly, usually within less than 15 minutes
Japanese version **did not** hang up after 10 hours. 
Tested in both MiST and MiSTer versions

jotego, 2020-01-01:
I have fixed this problem locally. I will soon update the RBF files.
Apparently the problem was related to the clock enable edge at which I was relasing the DTACK signal (pretty obscure indeed).

closed 2020-01-01 by jotego