jotego/jtbin #4: Sprite corruption F-1 Dream on 32MB xs1.1 sdram

Aquas77, 2019-12-29:
Video example: https://www.youtube.com/watch?v=Ik4ClQf7Ugs

jotego, 2019-12-30:
I have That XS v1.1 Sorgelig module RAM too. I can test with it. I am going to look into this strange bug. Thank you.

jotego, 2019-12-31:
Do you have to wait for a while for the issue to occur? If you just re-load the game ROM, does the problem go away -even if only for a while? After re-loading, how long it takes for it occur?
Sorry for the multiple questions but I need some hints to formulate a hypothesis.

Aquas77, 2019-12-31:
No I don't have to wait for the issue to occur. However, I have seen the title screen with the cars coming down have a graphic glitch. But I can't reproduce it at the moment, I had manually changed over to Mra format and now it doesn't occur, maybe that fixed that? 

I put a rom to load instead of rebooting the core to check your next question. Everything else as seen in the video is always present.  Reloading the rom does not fix the problem at all.  And it seems the same graphic glitches are always present, except for the aforementioned cars glitching on boot which I can't reproduce anymore.  The drivers pictures in the attract sequence are fine, as well as the title screen otherwise.

jotego, 2020-01-01:
Just copying here the video from #3. https://www.youtube.com/watch?v=jAOtcfRXaLg

jotego, 2020-01-03:
I have a hypothesis about this problem. Here are the technical details:

Tiger Road, F1-Dream and Bionic Commando are games that use a different sprite format in the ROM. After loading the ROM, the core reads, transforms and re-writes the sprites to the SDRAM so the same sprite engine used in other CAPCOM games can be used. If there are read/write problems in the SDRAM due to this, the sprites will appear corrupted.
Now, reading from the SDRAM seems ok as the games do work and there are no other issues. So there is probably something going wrong with the write. For some reason, the write in windbond memories is not working correctly. My write sequence does include a 20.8ns time between SDRAM activation and write (this time is called tRCD). I know Alliance -7 memories are specified to be no less than 21ns, so I am a bit out of spec for -7 memories. However, I think no Alliance user has complained about this. But maybe the Winbond memories require a longer tRCD time.

jotego, 2020-01-03:
I have a hypothesis about this problem. Here are the technical details:

Tiger Road, F1-Dream and Bionic Commando are games that use a different sprite format in the ROM. After loading the ROM, the core reads, transforms and re-writes the sprites to the SDRAM so the same sprite engine used in other CAPCOM games can be used. If there are read/write problems in the SDRAM due to this, the sprites will appear corrupted.
Now, reading from the SDRAM seems ok as the games do work and there are no other issues. So there is probably something going wrong with the write. For some reason, the write in windbond memories is not working correctly. My write sequence does include a 20.8ns time between SDRAM activation and write (this time is called tRCD). I know Alliance -7 memories are specified to be no less than 21ns, so I am a bit out of spec for -7 memories. However, I think no Alliance user has complained about this. But maybe the Winbond memories require a longer tRCD time.
I have asked Retroshop.pt to send me a Winbond memory to test. In the meantime, it would be helpful if you can give me the full code number written on the Winbond chip you use so I can check the exact data sheet of that device.
Other things you can try that may influence the memory are starting the core with a fresh cold memory and the opposite, starting it after having used the memory with a high frequency core, such as the NeoGeo one. The difference in memory temperature will likely affect the required time.
I do not want to just go and start making changes to the SDRAM controller without having a way to check the result so please bear with me while I wait for the memories.

shazz, 2020-01-03:
I have added some screenshots here:
[http://shazz.untergrund.net/mister/jotego/](url)

I'm using an XS 1.1 SDRAM board with a Winbond W9825G6KH-6 / 1823H / 67410K000ZV

Aquas77, 2020-01-04:
I did some tests with the suggestion of loading the neogeo core first and comparing versus resetting the MiSTer. The timestamps in the description of this video describe what each part is, https://www.youtube.com/watch?v=kd2fU0-1vCM

jotego, 2020-01-20:
Could you try again with today's Tiger Road/F1-Dream cores? the cores for 2020-1-20. I made a change to SDRAM timing that *may* have fixed this.
I'm still waiting for the Winbond SDRAM, by the way. It was sent over to me about a week ago so it should be here anytime soon.

Aquas77, 2020-01-22:
Still same corruption on my end.

onegoodsoul, 2020-02-23:
Same issue here. 32MB Winbond SDRAM.

jotego, 2020-02-27:
I have been working the last couple of days on addressing this issue. I think there's a good chance it will work now. I have:
-Decreased noise coupling on DQ lines
-Added SDRAM refresh during the ROM downloading process
Please run the update script so you get the 20200227 (today's) versions of Tiger Road and F1-Dream. Bionic Commando needs more work before an update due to a microcontroller problem.
I have not been able to check it because I did get a Winbond memory but it was defective. Nonetheless, I am rather confident on this. I am closing this issue. Please feel free to reopen it if the issue persists. Drop a comment if it worked ok.

Sorry for the delay. It wasn't obvious where the problem could be coming from. The CPS beta release gave me the hint I needed.

the1ross, 2020-03-07:
I just posted a new issue, then found this. I should have checked here first I guess. 

I have run the Mister update script but it doesn't seem to update all of your cores for some reason. I cannot find the new cores in the releases section either. Is there a different update script I should be running?

Thank you for all your hard work.

the1ross, 2020-03-07:
Ok, so I found your updater script. Updated everything. If anything, the games are even more corrupted now. 

[image]

jotego, 2020-03-07:
@the1ross please hit F12, the right and then go to About. Look at the version shown there. It should read **v20200305**
I finally got the Winbond module fixed so now I can verify this with four different memories. They all work. If it fails, please tell me the details of your memory module (manufacturer, size..) I am assuming it is a 32MB Winbond module.

jotego, 2020-03-07:
By the way, this only affects Bionic Commando, Tiger Road and F1-Dream because they need to re-structure the sprite ROM after loading it. It should be fixed for the later two. Bionic Commando has a problem with the MCU too and I am not updating it for now.

the1ross, 2020-03-09:
Yep, still garbled for me. I'm using a 32mb RAM from Ricardo. I don't think there's anything wrong with it because everything else works fine and I've done the RAM test. 
[image]

jotego, 2020-05-11:
I think there is only your case reported at the moment. So I am going to close this issue as a specific problem in your SDRAM module. I am sorry.

closed 2020-05-11 by jotego