jotego/jtbin #7: Bug - 1942 - Invisible bullets when HDMI powered off

thepulserifle, 2020-01-18:
Playing 1942 from VGA-->Component via the IO board, there appears to be an issue with the bullets that are fired from the small grey planes and the larger aircraft. Turning on the HDMI source lowers the behavior somewhat, but not entirely.

Steps to reproduce:
1) Unplug HDMI or power off any source that communicates with HDMI.
2) Start up 1942, coin up, and begin.
3) While playing 1942, focus on 100% accuracy (which will bring the planes down low).
4) Observe that you will occasionally explode without being hit by anything.

Using latest core: 20200103, latest MRA provided by the updates scripts/Bruno's INI updates. Using 128mb RAM.

Video of behavior: 
[Invisible Death 1](https://www.youtube.com/watch?v=I-UTwdUul3k)
[Invisible Death 2](https://www.youtube.com/watch?v=5n5SofRPtb4)

See also the MiSTer Facebook group for more details.

jotego, 2020-01-18:
Hi,

Thank you for reaching out to me about this.

I actually spent a few days these Christmass working on this core because some sprites were not being rendered leading to unseen collisions. But the version you refer to 20200103 includes the new sprite controller. I have just played to it again till level LAST 29 and didn't have any problem.
Could it be that the MRA is loading an older version of the core? Because this was really a problem for some mid term releases. You can check the core running by pressing F12 and look for the **about** menu item. Also check that your core is the official JT one. You can identify it because it has the JT logo as a watermark on the OSD menu.
HDMI should have no relationship to this at all. There is no internal signal relating the sprite controller with the status of the HDMI. However, if your board is not getting 5V correctly (because of a switch or splitter in the supply cable, or just a low supply source) then you may encounter odd behaviours.
So please, double check the RBF that is getting loaded and the voltage on your board.
I will close this for now.

jotego, 2020-01-18:
I think the sprite controller file got reverted to an older version when I merged a pull request from a collaborator. Let me dig for the right file and restore it.

jotego, 2020-01-21:
fixed

closed 2020-01-21 by jotego