jotego/jtbin #9: Black Tiger freeze sometimes.

Akaifutari, 2020-02-09:
Hi!
Sometimes, when I play Black Tiger's core the game freeze after the Game Over screen. It occurs sometimes, not always.
My MiSTer have 128mb.
I drop you a pic.

[image]

jotego, 2020-02-26:
I think this is the same as this is other [issue](https://github.com/jotego/jt_gng/issues/41). Please try today's update as I have improved the SDRAM robustness. I'll close this one and keep track on the other thread.

closed 2020-02-26 by jotego

jotego, 2020-02-27:
That's sad news. There is something else getting in the way of the Z80 then. Do you have the impression that the latest version is more robust or do you think it didn't make a difference?

Chocobobobo, 2020-02-27:
Let me hammer the latest rbf more, and I will post an update here after completing Black Tiger and Black Dragon a few more times.