jotego/jtbin #12: Gun.Smoke - Intro wanted poster display is skipped

thepulserifle, 2020-02-14:
The original arcade versions of Gun.Smoke had an opening that scrolls through all of the wanted posters in order. Currently, the Gun.Smoke core skips that intro.

Skip to :04 of this video if the timecode doesn't work to see what I am referring to:

https://www.youtube.com/watch?v=gZ4LUuk1kAc&t=4s

jotego, 2020-02-14:
You must be using a variation of the game ROM. The core does display the same content as the arcade when using the right ROM. Check out this side by side 10-minute video comparison.
https://youtu.be/zKvvTW3vvoY

closed 2020-02-14 by jotego

thepulserifle, 2020-02-14:
Disagree: please see linked videos:

https://youtu.be/4ctESnSNUDo

https://youtu.be/YqqNHUrLWew

jotego, 2020-02-14:
This is a software issue. Just use a different ROM.

thepulserifle, 2020-02-14:
I’ve tried every single rom and every single MRA. This problem existed
before the move to MRA. Can you show which rom actually shows the wanted
poster crawl?

I used the identical rom in MAME and your core during my test.