jotego/jtbin #14: CPS1 analog video output shifted right

paulb-nl, 2020-02-22:
The video on a 15Khz CRT is shifted right about 25 pixels

edru, 2020-02-23:
Same issue here.

paulb-nl, 2020-02-26:
Thanks for adjusting. Version 0226 is now shifted about 5 pixels too far to the left. Also it should be lowered about 4 lines.

paulb-nl, 2020-02-26:
Somewhat related: The internal video is shifted right by 2 pixels. There are 2 black pixels on the left in this screenshot made by the MiSTer screenshot feature.
[image]

jotego, 2020-03-02:
Thank you for that screenshot. It is a good indicator of the horizontal blanking being off.

paulb-nl, 2020-03-06:
Version 0306:
A little bit too far to the right now. Still needs to be lowered a few lines.
Hblank shift is fixed.

paulb-nl, 2020-04-18:
The vertical position is perfect with 0418 but the horizontal position was better centered with 0417.

jotego, 2020-05-02:
I think this is fixed now

closed 2020-05-02 by jotego

paulb-nl, 2020-05-02:
Thanks Jose. I recommend making Horizontal +5 the default 0 position because that is perfectly centered compared to consoles on the same CRT like SNES, Genesis and PC Engine. I also saw Andrew Moore use +5 in his latest CPS1 video.

paulb-nl, 2020-05-16:
With version 0515 it seems you have adjusted the horizontal position the wrong way. Now it needs more than +7 to center.