jotego/jtbin #20: CPS1 MRA alternative versions with problems

jotego, 2020-03-08:
Knights of the Round USA 911127 no sprites

M-Walrus, 2020-03-18:
I dont know if this is the right place for it, but Final Fight Anniversary Edition 3-player hack has messed up graphics on the "continue" screen. you can see a little bit of glitching for all characters, but Haggar is particularly bad
[image]

protogem2, 2020-03-28:
https://www.romhacking.net/forum/index.php?topic=28073.msg387239#msg387239

The hack does have some known issues, inherent to the rom. Might have to wait for the rom hackers to fix this.

houseoftheed, 2020-04-03:
The Area 88 alternate MRA files have distorted graphics after the bootup screen.

electricsheep26354, 2020-04-10:
Japanese versions of Street Fighter II World Warrior don't allow you to configure 6 buttons, only 4. tested all Japanese rom varients all seem to work bar this controller issue. Thanks

jotego, 2020-04-12:
All official ROMs work now in my development setup. I'll open a new issue for bootlegs and close this one.

closed 2020-04-12 by jotego