jotego/jtbin #32: CPS1 Fade to black seems wrong in Street Fighter II

loloC2C, 2020-04-10:
During changes between the character selection screen and the match-up presentation there is a fade to black that seems to be affecting the background layers only and not the sprites like on real hardware.

You can notice it in the below video (the countries flags are still displayed while the background is fading): https://youtu.be/Ole55lq7iPc

I think the same issue can be noticed in Final Fight too during the first level (I'll do a video capture later).

jotego, 2020-04-11:
That's actually how it should occur. You can also see in some transitions with the characters' eyes. The background will fade away and the eyes will still be visible for a split second. Original hardware behaves this way too.

closed 2020-04-11 by jotego

paulb-nl, 2020-04-11:
I think that the problem is that the fade out is not going as dark as the real hardware. This makes it more obvious.

Here is a comparison of the first frame that the background is visible when fading in. On the left is real hardware from https://www.youtube.com/watch?v=y1sY8dyxVpo
[image]

RetroDriven, 2020-04-11:
I don't have any comparison from my PCB yet but the fade does seem a little off to me just based on the amount I've played these games 😂

I'll try to get some comparison videos.

jotego, 2020-04-11:
@paulb-nl that's a very nice picture. Because of the way the bright control is implemented on the PCB; I think there is a large variation in how it looked, not only a PCB to PCB variation but also depending on the screen you connected to it. This is one place where customization by the user makes sense, to fit the memories of the system the user experienced.

But I don't think it has to do with this issue. I have clearly seen sprites left one frame on screen after the rest of the image was gone on the SF2 original PCB. The flags are drawn in the sprites layer (you can turn it on/off by pressing F10) so I think this is just the way the game worked.