jotego/jtbin #33: The King of Dragons, cymbal PCM seems very loud.

Left-Empty, 2020-04-10:
Greetings,

In the newest RBF (April 10th's), the cymbal rhythm sound effect in the King of Dragons seems disproportionally loud.
It is very apparent in one of the boss fight's music (which luckily happens to be the first and second bosses'), but in some other parts as well.
Thank you for your amazing work and best regards,

jotego, 2020-04-11:
Could you please enter the test menu of the game and look for the sound code of that cymbal?

Left-Empty, 2020-04-11:
I looked for it but sadly, there isn't a particular sound code for that sample. You can hear it easily by listening to sound 06 and 08, as it is very prominent in the intro of both those songs.

Left-Empty, 2020-05-01:
I listened attentively to compare with the videos I posted linked about in my other issue, and the sound of all PCM instruments seem to be too strong (they are very subdued in the video) : the cymbals just stand out more because they are likely the most noisy of the drums.
Meanwhile, the volume of the character's voice seem right, although they share the same PCM output I am guessing, as when a character is yelling, the drums are fading in the background.

jotego, 2020-05-06:
Some links to recordings:
https://discord.com/channels/558531469128237076/558531469128237078/707573400683675771
https://discord.com/channels/558531469128237076/558531469128237078/707573522528206898
https://discord.com/channels/558531469128237076/558531469128237078/707573718041493554

Left-Empty, 2020-05-09:
I checked with a few other games CPS-1 games, and while it isn't as striking, Knights of the Round seems to have a similar issue (on the first track of the first stage, the base drumline is quite proeminent versus the poor quality arcade footage I could find. The cymbal isn't as noisy though). In other games I'd be hard pressed to tell if there is a similar issue, as the drums are either way forward in the arcade (like Magic Sword), or way, way in the back (like Ghouls'n'Ghost).

jotego, 2020-05-09:
It looks like the ADPCM sounds too strong and too much HF content. 

It might sound too strong because a pipeline misalignment, or because the gain LUT is wrong, or because the multiplier is failing for some conditions.

The interpolator filter was commented out at some point during development. Its missing would produce the hiss noise.

I'll look into it.

jotego, 2020-05-10:
After looking at this all day, I haven't really found any thing broken. I have adjusted the volume balance a bit and I have enabled the interpolator for the ADPCM chip, which reduces the hiss.

Look at the attached file. It has the song in two tracks: one for ADPCM, one for FM and the mixed one. I can tell that the original recording has different equalization but I think the source sound is the same. I tend not to do much equalization because each user will have a different audio front end anyway so I cannot equalize that. But interpolator filters are a common feature I add, and that wasn't present before.

Please listen to the files and let me know what you think.
[sound.zip](https://github.com/jotego/jtbin/files/4606318/sound.zip)

Left-Empty, 2020-05-10:
Dang, I'm sorry for the red herring.
I have listened, but the PCM track is still very awkwardly distinct, while it just seems to merge into the background originally.
I tried fiddling with my amp's settings (I output the Mister to an amp then to my headphone on my setup), but the best I could get was muffling all the sounds, and then the PCM were still standing on top of the FM.

Thank you very much for looking into this.

Out of curiosity: lowering the PCM would lower all PCM effects sounds, including the yells, right? Yet their volumes are so distinct in the arcade game. And why is this only happening in this game! This is very puzzling!

Anyway, thank you again Jose.

jotego, 2020-05-10:
I think this is one of these hearing effects that things like MP3 are built on. For a while I actually thought there were more sounds at the same time but when I listened to the two tracks individually some sounds were not present.
If you know how to use [Audacity](https://audacity.es/) you can equalize each track individually. It is possible to characterize the frequency response of the original board but I am not sure how many users will actually care for the ammount of effort involved. And as I said above, users then will use their own equalizers anyway.
If my Patreon grows enough, I may have the resources to do that kind of high fidelity measurements eventually. But for now I think people may not be pleased if I get carried away with this sort of fine details.
It was good to add the interpolator anyway.
Thanks for reporting this.

jotego, 2020-05-10:
Oh, and yes. PCM/FM balance should be the same for all games as the A board is common to all.

TesterNes, 2022-11-11:
Is there any chance of revisiting this issue?
I don't know why it is closed when the problem persists. The music at stage 4 is a chaos because of this. 
Thanks.