jotego/jtbin #35: Street Fighter 2 Hyper Fighting Demo/Transition Screens

RetroDriven, 2020-04-10:
The Demo music cuts off right before the character select screen. It also seems to cut off after the Demo is done and looping through again. There seems to be some noticable artifacts when switching from different loading/transition screens. It's like it pauses briefly and cuts to the next transition.

RetroDriven, 2020-04-10:
The Demo music seems to be fine with Champion Edition and World Warrior but I do see the same delay in the transitions between screens there as well.

jotego, 2020-04-11:
Could you specify the exact MRA file where you saw this?

RetroDriven, 2020-04-11:
Street Fighter II'  Hyper Fighting -World 921209-.mra

This is the only MRA I can get working. The World MRA versions for all the SF2 games are the only ones working for me for some reason.

RetroDriven, 2020-04-11:
I can record a quick video with audio from the real PCB if you'd like. My PCB is the USA 921209.

RetroDriven, 2020-04-11:
Maybe this is what really happens for the World version and I'm just used to the USA version though.

Is the USA version working for you on your end?

RetroDriven, 2020-04-11:
https://youtu.be/qOmkjK56ccc

That's what I'm used to seeing so maybe it's just World vs USA and the cut off doesn't happen with USA and the Don't User Drugs screen.

RetroDriven, 2020-04-11:
After testing my PCB and the ROMs within MAME the music cut off is actually normal for the World version. Very interesting.

The black levels still seem more noticeable during the transition screens I think. I never noticed the graphics like Flags flashing like that until now. Another interesting thing you don't think about it notice until something like this happens in Mister. Pretty cool stuff in my opinion.

closed 2020-04-22 by jotego