jotego/jtbin #42: CPS1: The King of Dragons, inaccurate slowdowns?

Left-Empty, 2020-04-11:
Hello,

I noticed a slowdown that wasn't present in the arcade version (I think, as I don't have a PCB to test it): when the first boss launches the 3rd hit of his 3 hits combo (when he extends his chains to the max), there is an important slowdown that wasn't present originally I think.

If somebody could comment?

Cheers,

paulb-nl, 2020-04-11:
Here is footage of a PCB: https://youtu.be/Ajd0kXbGqhc?t=408

Left-Empty, 2020-04-11:
OH, didn't think of searching youtube, I feel so silly!

Indeed, there is a marked slowdown in the MiSTer version that don't seem to exist in the video.
Also, the game slows down a lot when you level up on the MiSTer, while it barely seemed to in the video.
But it's tough to judge still, as playing and watching games are such different things!

jotego, 2020-04-11:
I think this may be a version thing. USA910910 had that slowdown when leveling up whereas World 910711 didn't when I played them today.

Left-Empty, 2020-04-11:
Wow, you are absolutely right, and that's quite amazing. Even between different version of the Japanese version, all sharing the same date, the difference in the presence and/or length of slowdowns is quite staggering.
This is definitely something I hadn't even imagined!
Closing it as it seems it is the opposite of an issue now.

Left-Empty, 2020-04-18:
I repon the issue because I've found a very good footage of the Japanese PCB (at 60fps!)
https://www.youtube.com/watch?v=Frq9loZyF7Y

The game really slows down a lot in the April 17th build in parts where they are hardly any on display on the video. I understand someone else was posting about similar concerns with Ghouls'n'Ghosts. 
Maybe this video can provide hints?

jotego, 2020-04-18:
That's very interesting. Thanks a lot. I'm going to investigate it.

jotego, 2020-04-22:
After improving the accuracy of the DMA, I think it is ok now. I'll release it on Friday.

closed 2020-04-22 by jotego

Left-Empty, 2020-05-01:
Hello, I just tried the new release, and while the slowdowns seem still exaggerated at the default speed, they seem to be _perfectly_ accurate (from a sensible point of view, again) when turbo is activated!
I have no idea what this may mean but wanted to report.
Thanks for your work Jose!