jotego/jtbin #45: SF2 intro audience sprite issue

paulb-nl, 2020-04-12:
During the punch, parts of the body of the guy in the audience on the far right disappear.

[image]
[image]

jotego, 2020-04-13:
Could you confirm the version please?

paulb-nl, 2020-04-13:
jtcps1_20200410

Street Fighter II'  Champion Edition -World 920513-
Street Fighter II'  Hyper Fighting -World 921209-
Street Fighter II  The World Warrior -World 910522-

You can also see it in Smokemonsters video with all 3 games:
https://youtu.be/uTGqC1k9qYE?t=630

Chocobobobo, 2020-04-13:
At least, his hat is still around...

jotego, 2020-04-13:
I've seen it happening now. I'll get to it. Thank you :-)

jotego, 2020-04-23:
Fixed

closed 2020-04-23 by jotego