jotego/jtbin #46: Gun.Smoke flipped screen, sprites shifted down

Aquas77, 2020-04-13:
When I flip the screen with rotate off, the sprites are off-set and shifted down. Most noticeable with the snipers in the windows, immediately in Stage 1. But I believe this is affecting other sprites as well, barrels etc. This is not an issue it seems when the screen is not flipped... my TV only swivels in one direction :)

Flipped, no rotate picture: https://i.imgur.com/cj6nIki.png
No flip, no rotate picture (different scene but has window snipers): https://i.imgur.com/QTaYWTB.png

jotego, 2020-04-13:
Thank you. I hadn't noticed it.

The problem with 1943 and GunSmoke is that the original game does not support flipping the image, so there are some artifacts when flipping is forced upon the graphics hardware but the software side ignores it. You can see it at the bottom of the 1943 title screen too where a whole row stands out a bit.

If fixing this does not imply altering too much the hardware I'll do it.

jotego, 2020-05-09:
I think it should be possible to fix it by hardware. I'm changing it from enhancement to bug.

jotego, 2020-05-17:
Fixed in commit https://github.com/jotego/jt_gng/commit/c90a717080b5935f01263bf0608444734e83f5e7
Will update the binaries soon.

closed 2020-05-17 by jotego