jotego/jtbin #48: Credit button mapping issue on CPS1

Gerome91, 2020-04-14:
I have a problem configuring the controller mapping to launch a game (credit). The start button (credit) does not work it is confused with jotego credits

jotego, 2020-04-14:
I think this may happen if the MRA file did not have the button information correctly, including specific data in the rom index=1 section.
I will release corrected MRA files this Friday. Please verify it then and close this issue if it worked out.

0xc0ffea, 2020-04-15:
Sounds like I'm having the same button configuration issue that's making it impossible to play any of your cores, things were working fine.

I am using a usb 8 button arcade stick, joystick + 6 buttons + start +coin

The problem is that the credits menu seems to requires it be mapped to a button, I dont have a spare unused button and I can't seem to skip mapping a button for it, nor can I map it to a button on the attached usb keyboard.

So what happens is it somehow always ends up on the same button for coin insert .. which then blocks the coin. So I can watch the attract loop and view the credits. and that's it. This is happening on every single one of your cores (CPS1 and open source)

I'm not sure if this started with an update to Mister general or retrodriven (I tend to run them all sequentially)

Does the credits really need it's own hardware button ? Surely just having it as the first menu item would be sufficient

jotego, 2020-04-15:
I'm sorry about this. I think I made a mistake in yesterday's update for CPS0 cores which may have caused the issue. I am going to correct it today.

However, for CPS1 it should work for most MRA files. I actually use the same type of controller: 6 action buttons plus two extra. You can skip a button by pressing space. Is it the problem that you don't have a keyboard attached? There might be an alternative way to skip a definition without keyboard... I'll have to ask. That's part of MiSTer firmware, which is not my domain.

Credits can also be used to pause the game. This is enhanced in CPS1 because you can press 1P after pausing the game to disable the credits but still be in pause. Useful if you want to take a snapshot. I also moved the credits to the last position in the OSD in yesterday's update of CPS0 cores because it is already mapped to a button.

0xc0ffea, 2020-04-15:
I do have a keyboard attached, and have attempted to define a button for coin and skip credits with space .. I still end up with credits on the coin button.

jotego, 2020-04-15:
I have uploaded the changes for 1942, Vulgus, 1943, GunSmoke, GnG, Black Tiger and Commando. I'll get Tiger Road and F1-Dream done later.

0xc0ffea, 2020-04-16:
Just checked with GnG & Vulgus, that fixed it. thank you

jotego, 2020-04-17:
@0xc0ffea Tiger Road and F1-Dream are fixed now too.

jotego, 2020-04-22:
This was fixed for CPS last week.

closed 2020-04-22 by jotego