jotego/jtbin #49: CPS 1 - Bionic commando - wrong number of button

wwark, 2020-04-17:
Hello,

In the mra's of bionic commando there are 4 buttons (B0 to B3) but in the test mode there is just two buttons for this games.

Thanks,

jotego, 2020-04-23:
You're right about the test mode but I don't know what to do. Often arcade games do have input connections to the CPU that aren't apparently used. For Final Fight the extra button is marked as _cheat_. Maybe if I delete the buttons someone will come up saying that now they cannot perform some obscure hack.

Let me leave this open for a while and see if we can gather more opinions.

wwark, 2020-04-23:
ok don't know there is some extrat cheat button.
so it is based to mame regarding the number of buttons. The cheats is extra feature maybe not present on real arcade cabinet.

wwark, 2020-04-24:
I think maybe change the name of the button to label it for his purpose like BC1 (button cheat 1)

jotego, 2020-05-11:
I will publish the files following MAME database and let MRA editors modify them. Incorporating exceptions to all these cases in the MRA generation program is too much an overhead for me. I am counting on MRA editors to create many MRA variations, such as localised, with cheats on them, hacks, etc. So this can be another area for editors.

closed 2020-05-11 by jotego