jotego/jtbin #52: CPS Games around 15 o 20 left pixels  missing on CRT(vga out with scart)  on entire image

jorgalancomin, 2020-04-26:
Hi:

MIssing pixel on the entire image, the image look like bad centering , but when i enter in the service mode, i can't center the image because there is no more pixel at the left, by example chunli name in the vs title, barely i can't read the c and part of the h . 
This happend with all the cps games. 

Thanks.

jorgalancomin, 2020-04-26:
The version of core 'jtcps1_20200410.rbf' was fine, the problem is with the version 'jtcps1_20200417.rbf' .

paulb-nl, 2020-04-27:
jtcps1_20200410.rbf is much too far too the right so then the right side is cut off.
You should compare the left and right side of the life bars in Street fighter.

Chocobobobo, 2020-05-01:
Today's version includes H and V adjustments, you should tweak it as you wish now.

jotego, 2020-05-02:
I think people are happy now :-)

closed 2020-05-02 by jotego