jotego/jtbin #55: Double Dragon II - missing/garbled/blank screen output on core startup

mholzinger, 2020-04-26:
MiSTer ver: 191019
OS: Buildroot 2017.02
MRA: Double Dragon II - The Revenge.mra (md5 b5586d0f1201c17162a4aa831cc2bc7d)
Core: jtdd2_20200331.rbf (md5 d633568faa6aa31f2d1037e9c1bb1880)

Issue:
Core displays missing or garbled text on startup.

[image]

Workaround: 

1 - Display credits to enable sprite output

[image]

2 - Call in game reset from menu (F13)

[image]

3 - Display issues persist but are now visible

[image]

jotego, 2020-05-02:
Try resetting with F3 key after loading. It is a problem with the MCU very hard to track. Or maybe a problem with my glue logic and the reset. I have it in the back of my mind. For now just use F3.

jotego, 2020-05-21:
Summary of the situation here: https://github.com/jotego/jtdd/issues/8

closed 2020-05-21 by jotego