jotego/jtbin #56: Section Z bug with one enemy

Lmn-San, 2020-04-29:
I think I found a bug in Section Z. 
 When you shoot those little rotating radar on the ground (or on the ceiling) they should give points, and if you shoot really fast you should gain 10'000 points. I can not reproduce this behavior in your core.🙂
[https://www.youtube.com/watch?v=KOpTuV3EyKE&t=53s](url)

jotego, 2020-05-02:
Does the score go up? I wonder because if the score goes up but we don't see the points it will be a problem of leaving sprites out. I didn't know the original sprite limitation of the equipment and I used a reasonable one give the pixel clock.

Lmn-San, 2020-05-02:
Yes the score in your core go up. In fact the original game seems working like this, it let you shoot the radar and after a few seconds it add up the points to your score.
In your core it seems effectively that the sprites only are not showing.

jotego, 2020-05-06:
It's fixed now. It wasn't the sprite limit, which was correctly set but a bug in the sprite controller. Other CPS0 games might be affected too so I'll have to run a general update.
For now, you can just update to get SectionZ fixed.
Thank you very much for reporting this. It was really helpful.

closed 2020-05-06 by jotego