jotego/jtbin #57: cps1 Willow bug with 1/5/20 core

mdd45, 2020-05-01:
I have a bug to report with 1/5/2020 cps1 beta core with Willow:

[image]

Also there are gfx glitches with the bootlegs of Cadillacs and Dinosaurs and Punisher.

rarcos, 2020-05-01:
Cadillacs & Dinosaurs bootleg got fixed using this mapper in the MRA:

`<part> 25 40 88 73 FF </part>`

mdd45, 2020-05-01:
with this mapper i get a black screen, so still no for me.

jotego, 2020-05-01:
I'm uploading a file as soon as the compilation finishes. This has been caused by a corrupted file during compilation.

jotego, 2020-05-02:
The RBF file in the original attachment should work now correctly.

closed 2020-05-02 by jotego

mdd45, 2020-05-02:
thanks, now works!