jotego/jtbin #58: scandoubler set to 1 in MiSTer,

jotego, 2020-05-02:


paulb-nl, 2020-05-03:
I think direct_video should be 'norot' here:

https://github.com/jotego/jtframe/blob/333b0aa9dc31ef079d8f0de9cf2c0ed7432221c4/hdl/mister/sys/arcade_video.v#L192-L202

jotego, 2020-05-03:
Sorry, I don't understand what you mean by norot. Could you develop it a
bit, please?

paulb-nl, 2020-05-03:
norot is assigned above:
`norot       <= no_rotate | direct_video;`

so it becomes:
````verilog
always @(posedge VGA_CLK) begin
    VGA_CE <= norot       ? HDMI_CE : CE;
    if( norot       ? HDMI_CE : CE ) begin
        VGA_R  <= norot       ? HDMI_R  : R;
        VGA_G  <= norot       ? HDMI_G  : G;
        VGA_B  <= norot       ? HDMI_B  : B;
        VGA_HS <= norot       ? HDMI_HS : HS;
        VGA_VS <= norot       ? HDMI_VS : VS;
        VGA_DE <= norot       ? HDMI_DE : ~(HBL | VBL);
    end
end
````
So if it is not being rotated then it will output the scandoubled signals (HDMI_*) to VGA.

jotego, 2020-05-03:
I see, that is why direct_video set to zero disables the output. I will change it. Thank you!

jotego, 2020-05-08:
It is fixed now in JTFRAME. CPS0 cores got the update today. CPS1 and DD1/DD2 will get it soon.

jotego, 2020-05-09:
Fixed for CPS1 in commit https://github.com/jotego/jtcps1/commit/65328a1b156c1b7b0b1281448c34b32114c4b3b2

jotego, 2020-05-19:
Fixed for DD in commit https://github.com/jotego/jtbin/commit/a75c135e8acf4eba7e3303aef0b5ac7606fb5cc3

jotego, 2020-05-21:
This is solved for DD2 and Bionic Commando but the release is halted until these problems are solved:
https://github.com/jotego/jt_gng/issues/52
https://github.com/jotego/jtdd/issues/8

closed 2020-05-21 by jotego