jotego/jtbin #62: jtgng: unused dipswitch quite not unused

Left-Empty, 2020-05-08:
Greetings,

in today's MRA update, the last dipswitch of Ghosts'n'Goblins/Makaimura is labelled as unused, while it really isn't: it toggles invincibility. It wouldn't be much of an issue, if it wasn't for the fact the game defaults with it toggled on, which is quite confusing!

Cheers!

jotego, 2020-05-08:
Haha, isn't that a great twist! Many people will now thinking they suddenly
got really good at it!
I'll change it. Thank you!

Left-Empty, 2020-05-08:
> Many people will now thinking they suddenly got really good at it! I'll change it. 

I confess that was exactly my case, thinking I had an awful lot of luck, until I was stomped by the first boss ^.^;