jotego/jtbin #63: jtsectionz: screen still truncated with new display settings.

Left-Empty, 2020-05-09:
Hello,

Section Z is the only game I can't have a full picture displayed on my screen. Even at VPOS -8, the low end of the game doesn't touch the low border of the screen, and the top is cut out. This is the only game with that issue.

Cheers!

jotego, 2020-05-09:
I'm not sure I can solve this. That game uses overscan as far as I can tell. I'll make a test build doubling the effect of VPOS and we'll see if that helps.

paulb-nl, 2020-05-09:
Indeed the game uses overscan so it won't fit if your CRT is calibrated to hide overscan.
Unless you change your TV settings the top and bottom will be cut off when properly centered.

The credit text from the intro is above the overscan area so that should be visible at the bottom.

paulb-nl, 2020-05-09:
Looks like hblank starts too early. There is a pixel column missing on the right and a black pixel column on the left.

Left-Empty, 2020-05-10:
> Unless you change your TV settings the top and bottom will be cut off when properly centered.

You're absolutely right: I didn't notice the very small top cut (the bottom one being much bigger) but indeed, my screen isn't able to display it.

I'll leave Jose close the issue himself, as you posted another little issue and wouldn't want the comment going to the void.

jotego, 2020-05-10:
Thank you @Left-Empty
I'll close this issue. I've opened a new one for @paulb-nl 's observation: https://github.com/jotego/jt_gng/issues/50

closed 2020-05-10 by jotego