jotego/jtbin #64: CPS1 Missing text in Street Fighter Zero -CPS Changer, Brazil 950727 ?

Chocobobobo, 2020-05-11:
Everything checked on 0502 version of the core.
I don't own the original board to check, but it seems odd, so I am posting here, just  in case.
As you can below, the japanese version has explanations for manual mode and auto guard
[image]
While the brazilian version has an empty box (Auto has text in it)
[image]
Thanks for looking into it!

jotego, 2020-05-11:
I think they just didn't translate the text. It is not shown in MAME either.

closed 2020-05-11 by jotego

Chocobobobo, 2020-05-11:
That is super weird, because the text is translated for AUTO. Do you know anyone with the actual board to make it 100% sure? Thanks :)

jotego, 2020-05-11:
I don't. Maybe it's possible to buy a board from Japanese internet sellers.