jotego/jtbin #66: [CPS1] CRT H Adjust doesn't go as far left as before

houseoftheed, 2020-05-15:
The CRT H Adjust option in the 5/2 CPS1 core was able to go left far enough at +7 to match my original SF2HF board. The 5/15 CPS1 core has a gap at +7.

Screenshots:
5/2 core: https://drive.google.com/open?id=1elJGiB_BSxtZarXVE2a4gsfY0zS4l6yz
5/15 core: https://drive.google.com/open?id=1ecoVt2tLwXuAnDGxPAg5VHz-V9ijGXaF

yavimaya, 2020-05-15:
I only needed a CRT H adjust of +4 on v20200502 but now on v20200515 I need CRT H Adjust +7 and it is not as centered as before (maybe a +8 needed).

jotego, 2020-05-21:
I think this is fixed in commit https://github.com/jotego/jtcps1/commit/1f1889346d8097ea8a9a9d8d52cc245d30354aa7
You'll see it the next update

closed 2020-05-21 by jotego