jotego/jtbin #75: CPS1 Investigate MRA support for rare Capcom test build 'The World Warrior & Champion Edition'

mholzinger, 2020-05-22:
To investigate - Support for for rare Capcom test build 'The World Warrior & Champion Edition'

Source: Arcade projects user posts a new Street Fighter arcade build has been uncovered.

Post to the project and the forum are here: https://www.arcade-projects.com/forums/index.php?thread/12828-rare-game-the-world-warrior-champion-edition/

```Street Fighter II - The World Warrior &Champion Edition

Street Fighter 2 910522 - Etc. build

A-board 89626A-4 DASH
B-board 91634B-2
C-board PPU B-11```

- According to the forum post, this build may only work patched in ram and was a test build by capcom.

- Dump of files attached to this issue post.
[sf2-05222020.zip](https://github.com/jotego/jtbin/files/4670644/sf2-05222020.zip)

![sf2-910522-etc](https://user-images.githubusercontent.com/235958/82713818-35f0c400-9c41-11ea-99c3-ffbfb12c04e3.jpg)

- Also to note from Post- This may appear on a cloned Pang 3 board.

mholzinger, 2020-05-22:
Attaching tested working MRA to the issue. 

```<misterromdescription>
    <name>Street Fighter II' - Champion Edition</name>
    <setname>sf2wwce</setname>
    <year>1992</year>
    <manufacturer>Capcom</manufacturer>
    <rbf>jtcps1</rbf>
    <rom index="0" zip="sf2ce.zip|sf2-05222020.zip" md5="none">
        <!-- relative position of each ROM section in the file, discounting the header, in kilobytes -->
		<!-- Size of M68000 code 1024 kB -->
        <!-- Sound CPU size 64 kB -->
        <part>00 04 </part>
        <!-- OKI samples size 256 kB -->
        <part>40 04 </part>
        <!-- Graphics size 6144 kB -->
        <part>40 05 </part>
        <part repeat="10">FF</part>
		<!-- CPS-B config for sf2 -->
        <part> 32 41 FF FF FF FF 26 28 2A 2C 2E 36 00 30 08 10 20 00 </part>
        <!-- Mapper for sf2ce -->
        <part> 1F 40 C8 73 F7 </part>
		<part> 08 </part>
        <part repeat="24">FF</part>
        <!-- maincpu -->
        <interleave output="16">
            <part name="23" crc="ca331c3c" map="21"/>
		</interleave>
		 <interleave output="16">
            <part name="22" crc="e446d96a" map="21"/>
        </interleave>
        <!-- audiocpu -->
        <part name="9" crc="a4823a1b"/>
        <!-- oki -->
        <part name="18" crc="7f162009"/>
        <part name="19" crc="beade53f"/>
        <!-- gfx -->
        <interleave output="64">
            <part name="1" crc="22c9cc8e"  map="00000021"/>
			<part name="2" crc="57213be8"  map="00002100"/>
			<part name="3" crc="ba529b4f"  map="00210000"/>
			<part name="4" crc="4b1b33a8"  map="21000000"/>
         </interleave>
        <interleave output="64">
            <part name="5" crc="2c7e2229"  map="00000021"/>
			<part name="6" crc="b5548f17"  map="00002100"/>
			<part name="7" crc="14b84312"  map="00210000"/>
			<part name="8" crc="5e9cd89a" map="21000000"/>
		</interleave>
		<interleave output="64">
            <part name="10" crc="4267b696"  map="00000021"/>
			<part name="11" crc="df9b9146"  map="00002100"/>
			<part name="12" crc="a08a9786"  map="00210000"/>
			<part name="13" crc="80feac31"  map="21000000"/>
        </interleave>
    </rom>

    <!-- Set bit 1 for vertical games.
         Set bit 2 for games using 4-way joysticks -->
    <rom index="1"><part> 00 </part></rom>

    <buttons names="B0,B1,B2,B3,B4,B5,Start,Coin,Pause"
        default="A,B,X,Y,R,L,Select,Select,Start"/>
    <switches default="FF,FF,FF" base="8">
        <dip bits="0,2"	 name="Coin A" ids="4 Coins/1 Credit,3 Coins/1 Credit,2 Coins/1 Credit,1 Coin/6 Credits,1 Coin/4 Credits,1 Coin/3 Credits,1 Coin/2 Credits,1 Coin/1 Credit"/>
        <dip bits="3,5"	 name="Coin B" ids="4 Coins/1 Credit,3 Coins/1 Credit,2 Coins/1 Credit,1 Coin/6 Credits,1 Coin/4 Credits,1 Coin/3 Credits,1 Coin/2 Credits,1 Coin/1 Credit"/>
        <dip bits="6"	 name="2 to Start, 1 to Cont." ids="On,Off"/>
        <dip bits="8,10"	 name="Difficulty" ids="7 (Hardest),6,5,4,3 (Normal),2,1,0 (Easiest)"/>
        <dip bits="18"	 name="Free Play" ids="On,Off"/>
        <dip bits="19"	 name="Freeze" ids="On,Off"/>
        <dip bits="20"	 name="Flip Screen" ids="On,Off"/>
        <dip bits="21"	 name="Demo Sounds" ids="On,Off"/>
        <dip bits="22"	 name="Continue" ids="Yes,No"/>
        <dip bits="23"	 name="Game Mode" ids="Test,Game"/>
    </switches>
</misterromdescription>
```

[Street Fighter II The World Warrior Champion Edition -Etc 910522-.mra.zip](https://github.com/jotego/jtbin/files/4670674/Street.Fighter.II.The.World.Warrior.Champion.Edition.-Etc.910522-.mra.zip)

terminator2k2, 2020-05-23:
<dip bits="14,15" name="Region" ids="ETC,USA,JAPAN"/>