jotego/jtbin #76: No way to load roms in CPS1 for Mist

manast, 2020-05-24:
The core loads but only two menus are available, without any option load the roms.

Also the image is wobbling a bit, older versions did not suffer from this.

[image]
[image]

jotego, 2020-05-24:
You have to generate and use .ARC files using the mra tool. The best way is to copy the ARC, ROM and RBF files to a folder. Then set the system attribute of that folder. That way MiST will show it.
I'll upload a version with the wobbling fixed in a few minutes. Stay tuned.

jotego, 2020-05-24:
I won't be able to upload the binary today. Use the current one for now.

manast, 2020-05-24:
yes, I improved mra_tool to easily generate the files so I kind of know what I am doing :) but I did not know the .rbf needs to be in the same directory as the rom files. thanks.

jotego, 2020-05-24:
The new binary is there now. I found some time to do it. This corrects the horizontal sync length to match the expected 4.5us. Everything should be fine now.

closed 2020-05-24 by jotego

manast, 2020-05-24:
The wobbling is gone, however, I did as you said, put all the ROM + ARC files in the root directory (gets quite messy btw), then same problem as before, no option to load roms and the Scandoubler FX is gone also which I guess it is ok since this is a standard CRT.

jotego, 2020-05-24:
It doesn't need to be the root folder. It can be a subfolder but you need to set the system attribute of that folder. There is no option to load ROMs. You have to select the ARC file and it will load both the RBF and the ROM.
No Scandoubler FX for MiST to save space in the device.

manast, 2020-05-24:
ah ok, sorry, I am confused since this is different from previous versions where first we should load the core then load the rom. So by loading the ARC it seems it loads the core correctly, the menu looks good, however the game will not boot, just black screen. I am testing with several and is the same with all of them. I used the latest MRAs from jtbin repo to generate the rom files and arc files with same procedure as before.

manast, 2020-05-24:
I upgraded to latest mist firmware and now it works :).

manast, 2020-05-24:
I wrote too soon... the games boot but they go directly to test screen and it is displayed upside down.

manast, 2020-05-24:
and the F12 OSD menu does not work seems like the core just crashed.

jotego, 2020-05-24:
The MRA tool generates lines too long for MiST and corrupt the OSD menu. I think they're working on it. For a while the tool didn't do that and options were ok but something went wrong lately.
You can delete long lines from the ARC file but doing that for 200 files is too much.
Just do it for the titles you care about and wait for the MRA update for the rest

Change test and flip in DIP settings. Save settings later. You may need to delete old CFG files.

jotego, 2020-05-24:
Get the [right ARC files from here](https://github.com/jotego/jtbin/tree/master/mist/cps1/releases).