jotego/jtbin #79: CPS0 - Bionic Commando not using new framework

mholzinger, 2020-05-27:
RBF file - jtbiocom_20191201.rbf expects folder in path: `/Games/BIONICC/`

Uses older format rom file `JTBIOCOM.rom` and Ignores `/Games/mame/bionicc.zip`

```$git log ./mister/biocom/releases/jtbiocom_20191201.rbf
commit 6b78de96b1f40617b3fcd59558e261374cbaab85
Author: Jose Tejada <jotego@gmail.com>
Date:   Sun Dec 1 21:41:04 2019 +0100
```

lomdar67, 2020-05-27:
Can't confirm this. I have all my roms ONLY in games/mame/, which is actually a cifs file share and the jtbiocom_20191201.rbf finds the bionicc.zip there.

jotego, 2020-05-27:
That has nothing to do with the RBF file. RBF knows nothing about the filesystem. It is just a binary file that programs the internal switches inside the FPGA.
Your issue must be somewhere else.

closed 2020-05-27 by jotego