jotego/jtbin #83: update_jtcores.sh downloads popey core to _Arcades rather than cores

MAME-GETTER, 2020-06-04:
```
Checking popeye
URL: https://github.com/jotego/jtbin/tree/master/mister/popeye
Downloading jtpopeye_20191227.rbf to /media/fat/_Arcade/jtpopeye_20191227.rbf
URL: https://github.com/jotego/jtbin/blob/master/mister/popeye/releases/jtpopeye_20191227.rbf?raw=true
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   153  100   153    0     0    466      0 --:--:-- --:--:-- --:--:--   495
100   164  100   164    0     0    316      0 --:--:-- --:--:-- --:--:-- 23428
100 2776k  100 2776k    0     0  1341k      0  0:00:02  0:00:02 --:--:-- 3795k
Deleting old jtpopeye files
```

theypsilon, 2020-06-04:
That's true, it should get fixed as soon as there is a Popeye MRA in its releases folder.

MJY71, 2020-06-05:
There are MRAs in the jtpopeye zip in the upper level directory of that repo that could be moved to the releases folder.

theypsilon, 2020-06-05:
Good to know, I will open a PR with that.

jotego, 2020-06-05:
The MRA is in place now thanks to @theypsilon. It should work now.

closed 2020-06-05 by jotego