jotego/jtbin #84: update_jtcores.sh MRAs _Alternatives

MAME-GETTER, 2020-06-04:
The update script should not force downloading of _alternatives.

The official MiSTer update does not download _alternatives nither should the update_jtcores.sh

This could be a switch in the ini file. 

Why does the update script place a file at:
```
/root/fat/Scripts/.mister_updater_jt# pwd
/root/fat/Scripts/.mister_updater_jt
/root/fat/Scripts/.mister_updater_jt# ls -l
total 0
-rwxr-xr-x 1 root root 0 Jun  4 08:34 MRA-Alternatives_20200604.zip
```
The update script should seek to minimize writes to the sd card.  I would be better if these were written to /tmp as /tmp is a ram disk.

mtime on the MRA file should be enough to see if needs to be updates/

theypsilon, 2020-06-04:
Snce the version 4.0.3 of the official Updater, the default value of MAME_ALT_ROMS is set to true.

Here is the changelog:
# Version 4.0.3 - 2020-02-24 - Changed MAME_ARCADE_ROMS and MAME_ALT_ROMS default value to "true"; added _Other core directory and removed Arduboy from SD root; renamed CORE_CATEGORY_PATHS["cores"] to CORE_CATEGORY_PATHS["computer-cores"] for better readibility, "cores" still works for both CORE_CATEGORY_PATHS and filters; code clean up by frederic-mahe (thank you very much).

About the writing files part. As of know, this script is mimicking the behavior of the official updater on MRA-Alternatives handling. In fact, since this repository is a direct fork from the official updater and there are just a few changes, I haven't touched that part, it's exactly the same implementation.

I think your feedback is good, and maybe we could address it to the official updater repository at https://github.com/MiSTer-devel/Updater_script_MiSTer . If some mechanism as the one you suggest  gets implemented there, this repo may inherit it very easily. But for know, I think is better to not deviate from their way of doing things, so we can stay on sink for the most part and people can expect same behaviours apply to both scripts (the official one, and the jotego one).

MAME-GETTER, 2020-06-05:
I guess this will fix what I want, which is not to have the _alternatives directory. 

```
/media/fat/Scripts/.mister_updater_jt# ls -lrt
total 0
-rwxr-xr-x 1 root root 0 Jun  4 08:34 MRA-Alternatives_99999999.zip

```

theypsilon, 2020-06-05:
There is another way to do it. You can create a update_jtcores.ini file in the Scripts folder with this line:

`MAME_ALT_ROMS="false"`

Same thing with the MiSTer-devel updater, if you want to avoid those MRA-Alternatives also.

MAME-GETTER, 2020-06-12:
closing 

`MAME_ALT_ROMS="false"
`
works well.

closed 2020-06-27 by jotego