jotego/jtbin #85: update_jtcores.sh has bash errors

MAME-GETTER, 2020-06-04:
```
/root/fat/Scripts# ./update_jtcores.sh
Downloading and executing
mister_updater_jtcores.sh

Downloading and executing
mister_updater.sh

|ooooooooooooooooooooooooooooooooo+++++++++++oooooooooooooooooooooooooooooooooo|
|oooooooooooooooooooooooooooooooo+. .    . .+oooooooooooooooooooooooooooooooooo|
|oooooooooooooooooooooooooooooooo~         :o++oooooooooooooooooooooooooooooooo|
|ooooooooooooooooooooooooooo+ooo+.        .++.:oo+oo+oooooo+o+oo+oooooo++oooooo|
|oooooooooooooooooooooooo+.......          .. .............................:ooo|
|ooooooooooooooooooooooo+.                                                .+o+o|
|ooooooooooooooooooooooo:                                                 :o:.+|
|oooooooooooooooooooooo+.                                                .++.:o|
|oooooooooooooooooooooo:.....           ...........          ....... ....:o~.+o|
|ooooooooooooooooooooooo+++++~         ~+++:++:++++.         ++++++++++++++.+oo|
|ooooooooooooooooooooooooo+:~         .++.~:::::::.         .o+.~:::::::::::+oo|
|ooooooooooooooooooooooooooo:         :o~.+oooooo+.         ++.~ooooooooooooooo|
|oo~........~oooooooooooooo+.        .++.:ooooooo+         ~o:.+ooooooooooooooo|
|o:         ~o++ooooooooooo.         +o~.ooooooo+.        .++.~oooooooooooooooo|
|o.         :+.:ooooooooo+.         ~o+.+ooooooo:         ~o:.+oooooooooooooooo|
|o.         ...~:::::::..          .++.~ooooooo+.        .o+.:ooooooooooooooooo|
|o~                               ~o+..+ooooooo~         +o~.+ooooooooooooooooo|
|o+.                           .~+o+..+ooooooo+.        .o+.+oooooooooooooooooo|
|ooo~.                       .:+o+..:+oooooooo.         +o..ooooooooooooooooooo|
|oooo+:...              ..~:+++:..:+ooooooooo+         .o+.+ooooooooooooooooooo|
|oooooo+++::::::::::++++++++~~.~++ooooooooooo+:+:::::+:++.~oooooooooooooooooooo|
|oooooooo+++:::::::::~:~~~~:++oooooooooooooooooo+::::~::~.+oooooooooooooooooooo|

Original script by Alessandro "Locutus73" Miele

MiSTer Updater version 4.0.7

basename: invalid option -- '.'
Try 'basename --help' for more information.
Performing a full updater resync

Checking 1942
URL: https://github.com/jotego/jtbin/tree/master/mister/1942
Nothing to update
Checking MRA files

Checking 1943
URL: https://github.com/jotego/jtbin/tree/master/mister/1943
Nothing to update
Checking MRA files

Checking biocom
URL: https://github.com/jotego/jtbin/tree/master/mister/biocom
Nothing to update
Checking MRA files

Checking btiger
URL: https://github.com/jotego/jtbin/tree/master/mister/btiger
Nothing to update
Checking MRA files

Checking commando
URL: https://github.com/jotego/jtbin/tree/master/mister/commando
Nothing to update
Checking MRA files

Checking cps1
URL: https://github.com/jotego/jtbin/tree/master/mister/cps1
Nothing to update
Checking MRA files

Checking dd
URL: https://github.com/jotego/jtbin/tree/master/mister/dd
Nothing to update
Checking MRA files

Checking dd2
URL: https://github.com/jotego/jtbin/tree/master/mister/dd2
Nothing to update
Checking MRA files

Checking f1dream
URL: https://github.com/jotego/jtbin/tree/master/mister/f1dream
Nothing to update
Checking MRA files

Checking gng
URL: https://github.com/jotego/jtbin/tree/master/mister/gng
Nothing to update
Checking MRA files

Checking gunsmoke
URL: https://github.com/jotego/jtbin/tree/master/mister/gunsmoke
Nothing to update
Checking MRA files

Checking popeye
URL: https://github.com/jotego/jtbin/tree/master/mister/popeye
Nothing to update

Checking tora
URL: https://github.com/jotego/jtbin/tree/master/mister/tora
Nothing to update
Checking MRA files

Checking sectionz
URL: https://github.com/jotego/jtbin/tree/master/mister/sectionz
Nothing to update
Checking MRA files

Checking vulgus
URL: https://github.com/jotego/jtbin/tree/master/mister/vulgus
Nothing to update
Checking MRA files

Checking MRA-Alternatives
URL: https://github.com/jotego/jtbin/mister/MRA-Alternatives_MiSTer
Nothing to update

basename: invalid option -- '.'
Try 'basename --help' for more information.
bash: line 1168: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1169: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1170: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1171: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1176: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1179: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1180: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1186: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1189: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1190: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1195: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1198: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1199: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1205: /media/fat/Scripts/.mister_updater_jt/: Is a directory
Updater log, see /media/fat/Scripts/.mister_updater_jt/
===========================
cat: /media/fat/Scripts/.mister_updater_jt/: Is a directory
===========================

bash: line 1223: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1225: /media/fat/Scripts/.mister_updater_jt/: Is a directory
bash: line 1226: /media/fat/Scripts/.mister_updater_jt/: Is a directory
Done!
```

theypsilon, 2020-06-05:
Thanks a lot for reporting @MAME-GETTER, that bug should be fixed now.

jotego, 2020-06-05:
I close it for now. @MAME-GETTER please reopen it if you still see the issue

closed 2020-06-05 by jotego