jotego/jtbin #90: Ghosts'n'Goblins - sideways scrolling not smooth

dfilskov, 2020-06-08:
It seems the background doesn't scroll smoothly. Perhaps it's like scroll 1 frame, then pause 1 frame, scroll 1 frame, pause 1 frame etc. The quick level oveview scrolls nicely but ingame when running the background seems jumpy.

Is the original hardware like that too? - if not perhaps the scrollling could be smooth here too?

lroby74, 2020-06-08:
On a CRT or HDMI?

dfilskov, 2020-06-08:
On HDMI with integer scaling and lancos2_15 (I think that's what I'm using but I can check)

lroby74, 2020-06-08:
i saw a big difference on scrolling on several games between HDMI and CRT, on CRT was perfect, on HDMI not

jotego, 2020-06-08:
Hi @dfilskov, I'm afraid @lroby74 is right. This is likely an effect of the upscaling. I can only tell you to try a different screen filter.
The reason why this is most noticeable during gameplay than during the map is that the game doesn't seem to use 1-pixel constant speed scrolling during gameplay whereas it uses it for the map. Note that this is the same hardware used in other CAPCOM pre-CPS titles which show a nice scroll so it boils down to a software design issue on the original game.
I'll leave the issue open for a couple of days in case you want to give us feedback about a specific filter that works well. Maybe a filter that averages with an attenuated version of the previous frame would work well.

dfilskov, 2020-06-08:
Thank you, guys, for the explanation and for leaving the issue open a bit longer. I'm trying to **understand** the technical aspect here.

I have tried lots of different filters and come to the conclusion **that lancons2_15 is the best** for scrolling. But it doesn't help in this case (neither for Wonder Boy btw - which I also reported and Mr-X wrote that the reason was known and that it would be corrected later).

I can set MiSTer to NOT scale up at all (currently I'm using **integer scaling** (vertically), though, to get the best smooth scrolling (up/down) and it works perfectly for other cores)

However ... unfortunately MiSTer _enforces_ sideways stretching of pixels (blurring the image a bit) to match the aspect ratio of the original analogue CRT screen. **lancos2_15**, though, should fix all shimmering - and it does on e.g. C64 Giana Sisters which scrolls sideways perfectly even upscaled without integer!

So what's different about the GnG scrolling on HDMI in, say, 50 Hz (when displayed without scaling) and CRT the same 50 Hz framerate? - wouldn't that stutter? - or would it still stutter through HDMI? - and if so, why?

lroby74, 2020-06-08:
"So what's different about the GnG scrolling? - would it NOT stutter on HDMI in, say, 50 Hz (when displayed without scaling) and CRT in the same 50 Hz framerate? - or would it still stutter through HDMI? - and if so, why?"  

Yes on HDMI result will be always worse because technology is completely different from the CRT, which if I remember correctly, has a slight delay in turning off the pixels and this causes the nuance that generates a smooth scrolling

jotego, 2020-06-08:
I agree with @lroby74 and if like to stress that GnG doesn't seem to advance the scroll variable by a constant value per frame, say 2 pixels per frame. I think that odd speed doesn't look nice in our crispy modern screens.
I'd like to experiment one day with a filter that takes into account the previous frame. I don't think MiSTer can do that right now though.

dfilskov, 2020-06-08:
Thanks again, guys! - I assume you've seen GnG run perfectly smooth ingame using the VGA on th IO board.
But can you also make GnG run smoothly on a CRT using Direct Video? (via HDMI) - that way we can rule out that it's the scaling or the flatpanel tech ... I mean IF it stutters on a CRT using Direct Video something else must be wrong, right? :) - perhaps that could be worth a test?

dfilskov, 2020-06-09:
Btw this issue is not present ingame in **Ghouls**'n'Ghosts (parallax scrolling). Its overview scrolls less smooth than **Ghosts**'nG's overview and there's a bit of shimmering even with the right filter but it's still as smoot as it can get on an flatpanel when not being pixel perfect - but somehow **Ghouls**'n'Ghosts scrolls smoothly ingame even when Ghost'n'Goblins doesn't. Perhaps that can help for comparison.

jegayon, 2020-06-26:
> 
> 
> Thanks again, guys! - I assume you've seen GnG run perfectly smooth ingame using the VGA on the IO board.
> But can you also make GnG run smoothly on a CRT using Direct Video? 

Hi. GnG scroll can't run smoothly like Ghouls'n'Ghosts becase the game don't move background one pixel (or two, or three...) each frame and cause this scroll stuttering. It's not an hardware issue, just an software issue. :)

dfilskov, 2020-06-27:
That's so odd :) does it stutter on CRT too?

Or do you know the "science" - the explanation - of why evenly moving backgrounds (moving say, 5 pixels each frame) look cool and smooth on a crt while stuttering on a flat-panal running the same refresh rate?

jotego, 2020-06-27:
They look fine on CRT because the CRT pixels glow for more than a frame producing a low pass filter effect.

closed 2020-06-27 by jotego

dfilskov, 2020-06-27:
Thanks, I see. I'm really curious about why the glow/filter effece tricks the brain - and why the flat-panel does not trick the brain.

dfilskov, 2020-06-27:
I know a professor - a scentist - who works with exactly this kind of cognitive perception. I'll ask him but do guys know exactly how the GnG background moves? - do you know someone who can record it perfectly without frameloss?

paulb-nl, 2020-09-09:
The game also stutters on CRT. It is just inherent to the game.

Also CRTs are low persistence displays so pixels glow only 1-2 milliseconds. This is the reason CRTs don't have motion blur and LCDs do.

dfilskov, 2020-09-09:
Oh - perhaps it's inherent ... here's the real thing - can you tell?

https://www.youtube.com/watch?v=5ENZcJTk9bw

dfilskov, 2020-09-09:
Much better video here: https://www.youtube.com/watch?v=h7473QQwZYA

jotego, 2020-09-09:
The platforms are not smooth but those are rendered as sprites. So that's definetely a software problem in the original.
But, is the scroll ok in the original board? The mountains look in the video to show the same jumpy effect as seen in the FPGA core.

[image]

Pay attention to them. I do think the original is not smooth although I think it is more noticeable in modern screens.

dfilskov, 2020-09-10:
If the same issues are in the original then the FPGA conversion is perfect :)

dfilskov, 2020-09-10:
Perhaps this scrolling at varying frame-rates test can tell us something:

https://www.testufo.com/eyetracking#pattern=stars&speed=1