jotego/jtbin #92: Fire opens the OSD if used in a menu combo

dfilskov, 2020-06-09:
For some users most JT cores open the menu instead of firing when you press the fire button :)

The Fire button (aka A button aka B0 button) opens the Menu if the fire button is used as the second button in a 2-button combo for opening the Menu / the OSD.

It only seems to happen in JT cores.

jotego, 2020-06-10:
This is a MiSTer configuration thing. Go to the main MiSTer menu and define the combo button for the OSD different. This cannot be controlled from cores and it is independent of cores.
Please have a look at the MiSTer wiki if you cannot find the place to define it.

closed 2020-06-10 by jotego

dfilskov, 2020-06-10:
Thanks, Jose! - I know, I can change the combo to avoid the issue. But it's not an issue in the other cores - two button combos (e.g. Start+Fire to open OSD) works perfectly in all cores except from JT cores afaik.

So perhaps JT cores are handling input differently than most other cores and could be updated to make it work?

The fire button press should not be registeret in the game when used in the opens menu combo, right?

In the JT cores this also causes the menu's joystick mapping of keys to close instantly when you click it. I'll make a video to show you what I mean :)

jotego, 2020-06-10:
I am a bit skeptical about this, but please do share your video.

dfilskov, 2020-06-10:
I understand - thank you for still being open to any possible relevant knowledge from my video! - I'll have it ready today or tomorrow.