jotego/jtbin #93: CPS1 cores are not playable with DB15 joystick anymore

luco59, 2020-06-12:
Hi,

To play with your CP1 cores I use the Antonio Villena’s Mister with DB9 port :) A really nice piece of hardware :)

Natively all the CPS1 cores were natively playable with a DB15 Neo Geo pad (with a DB9 to DB15 splitter) by simply selecting ‘SNAC : 1 player’.

Recently my DB15 has begun to stop functionning.

In your opinion, is this issue  could be a consequence of an update on your side or a would It be a consequence of an update of the SNAC support ? I have remarked that SNAC is now usable with the MegaCD and SNES cores for example.

Thank you for your help.

Your work is really impressive 👌

Best regards,

Luc (patreon mail : jolyluc @ gmail.com)

jotego, 2020-06-27:
Hi Luc,
Sorry for the delay in response here. I talked about this on other places and then forgot about to write it here. The good thing about issues is that they stay there! So here I am.

When updating the MiSTer framework I must have broken the DB15 support. I have since received sample code to support DB9 and DB15 and I did spend one day in refactoring things to make it fit in but I need to spend another day before releasing an update. As the MiSTer framework had an update this week, I will work on the update and DB9/15 support soon. Probably the second week of July.

luco59, 2020-06-27:
Hi Jose,

Thank you 😊 DB9 users will be very happy 🎮

Regards,

Luc

jotego, 2021-02-20:
I have spent several days looking at this. I have decided not to support the following DB9 controllers in the current board:
-Megadrive
-Turbografx
-SNES
-NES

The NeoGeo controller is still on the table. I need to spend another day or two verifying it before adding it to public releases.

luco59, 2021-02-21:
Thank you for the news.

DB15 support would be great ;)

jotego, 2021-11-20:
This was fixed a while ago

closed 2021-11-20 by jotego