jotego/jtbin #96: Not enough Bubbles in Bubble Bobble intro

Chocobobobo, 2020-06-13:
First of all, this bug is courtesy of Dampfwalze.
During the intro, you have bubbles moving around in circle motion in the original, but only 2 bubbles going up vertically in the current version of the core.
Proper behavior:
https://youtu.be/1dadJO4YtMk
[image]

Current MiSTer behavior
https://youtu.be/n6UfKNzD0kI
[image]

Jotego mentioned it might be MCU related, but we will let the pro take it from there! :)

jotego, 2020-06-16:
Simulation of the video circuit with memory dumps from MAME shows the correct behaviour. So this seems to be a problem on the CPU side of the game. I will review the memory connections. I hope this is not a bug in the 6801 MCU softcore.

[image]

jotego, 2020-06-27:
Update: I tried several CPU/Sub CPU/MCU speed combinations in MAME just to see if the problem could be related to the MCU speed but they all showed bubbles ok.