jotego/jtbin #97: SF2WW USA 911101 Flicker

solo2081, 2020-06-14:
I check my ram and it seems ok pass all test . this flicker happens in all stages but easy to see in sagat stage . when you beat a fighter in this case sagat with a perfect hold the
 buttons to increase the speed of the score count it will cause some parts of the background to flicker in video look at the rock and sagat body on the ground video is in zip file

[IMG_3680[3444].zip](https://github.com/jotego/jtbin/files/4775746/IMG_3680.3444.zip)

paulb-nl, 2020-06-14:
Yes on the Sagat stage near the tree that happens with the arcade pcb.

Here is a youtube video where the same thing happens: [https://youtu.be/2-u1h_lF53E?t=298](https://youtu.be/2-u1h_lF53E?t=298)

mikey1964, 2020-06-14:
good find paulb-nl looks the same