jotego/jtbin #103: Street Fighter Alpha Publicity USA - Bison corner graphic glitch

Aquas77, 2020-06-26:
When Bison is hit in the left corner, occasionally his sprite disappears completely for a few frames. I've only observed this interaction in the left corner and with Bison, interacting with him against the corner usually.

This clip demonstrates the sprite disappearing, about 18 seconds into the clip, or at 87 seconds of Round 2 in game time when Charlie does a flash kick hitting Bison behind him, it happens quickly:

https://clips.twitch.tv/ReliableFrozenEagleBuddhaBar

I've asked a few other knowledgeable people of the game and they had never seen something like that before. I have also had one person try to recreate the glitch in MAME, without success.

jotego, 2020-06-26:
Yes, I can see the problem. Thank you for spotting it. This would probably be easy to debug if only could I get a memory dump from MAME at that point. I'll try playing in MAME and saving when Bison is hit in the left corner to see if I can then reproduce the problem when getting that memory dump through the video engine in simulation.

If you happen to find a repeatable pattern that produces the problem that will help a lot.

[image]

Aquas77, 2020-06-27:
Okay! Thankfully it turns out its easy to reproduce. Put Bison standing in the left corner and hit him a flash kick from Charlie. It also occurs with Ryu's dragon punch, perhaps even other similar moves that knock the opponent into the air I guess?

You'll have to go into Versus mode and select Bison and Charlie/Ryu/etc.

Select Bison through a code. Hold Start on the Character Selection screen and hover over Random Select then put in this input.

If you are 1P : 
        Bison - down, down, left, left, down, left, left
If you are 2P :
        Bison - down, down, right, right, down, right, right

Then press Light Punch + Fierce Punch together, after the inputs to activate Bison.

Good luck