jotego/jtbin #105: CPS1 background layer missed in Three Wonders intro.

jegayon, 2020-06-28:
There seems to be a problem in the background layer in the introduction of Three Wonders:

[image]
[image]
[image]
I hope these screenshots help.

jotego, 2020-06-29:
What's the problem? Which version of the game are you using?

jegayon, 2020-06-29:
Sorry!
-Rom version is Three Wonders (World 910520). MIST Core version jtcps1_20200524.
-The problem is missed layer in background. When plays this into, the background apears with two layer scroll parallax but in FPGA only shows front layer.

paulb-nl, 2020-09-29:
I still see this issue with core version 20200910 on MiSTer

jotego, 2020-09-29:
I don't get it correctly now either. The CPS-B configuration data must be wrong, either in the MRA file or the ROM mapper that gets compiled into the RBF file. I'll fix it.