jotego/jtbin #109: MRAs are duplicated in the repo, that could be confusing to some people

theypsilon, 2020-06-30:
@gyurco @jotego 

I noticed this: https://github.com/jotego/jtbin/pull/100, and there is a problem with this PR. It won't get downloaded by the Updater, because it is in the wrong folder. MRAs should be in each releases folder of these subdirectories in order to get noticed by the Updater: https://github.com/jotego/jtbin/tree/master/mister

I suggest to delete that mra folder in order to avoid confussion.

About the _alts, they are in a zip. I agree that is ugly, but it's the way the official updater handles that. We could consider auto generating zip files on push to any MRA-Alternative if that makes sense, but for that I think is still needed to stick to the official MRA-Alternatives folder structure. Which is like:

_alternatives/_Double Dragon/alt MRAs...
_alternatives/_Double Dragon 2/alt MRAs...
_alternatives/_Street Fighter II/alt MRAs...

theypsilon, 2020-06-30:
About @gyurco change I think I could try to code a migration to move his changes the releases folder if that helps.

gyurco, 2020-06-30:
Where is this releases folder? Why the MiST/SiDi/etc should get the necessary files from the MiSTer folder? It's like MiSTer wants to rule everything.

gyurco, 2020-06-30:
Btw, the above PR is not relevant for MiSTer, it has a hardcoded DIP page.

theypsilon, 2020-06-30:
@gyurco I think we can try to find a solution that is more system-agnostic.

The problem about modifying the Updater for MiSTer is that it would be a very substantial change make, and I'm trying to keep the Updater fork as minimal as possible, so it stays compatible uppon new developments, it has feature-parity, inherits fixes from upstream and all of that with very low maintenance.

We could have an automated task that syncs MRAs with the same name to the latest version. That way, it doesn't matter where did you update the MRA, because a bot will put in all places the same changes. Does it sound good to you @gyurco ?

gyurco, 2020-06-30:
As it is possible to store symlinks in git repos, that wouldn't be better than syncing? (At the end, it's jotego's repo, I think he should decide).

theypsilon, 2020-06-30:
About alts, things are a bit different, since in MiSTer they are grouped by game. Here is an example of that structure: https://github.com/theypsilon/jtbin/tree/alts-demo/mra/_alternatives

I think I could make a solution that would automatically zip a folder if any change is detected and commit that zip into this path: https://github.com/jotego/jtbin/tree/master/mister/MRA-Alternatives_MiSTer/releases

But for that, I think that at least they should be grouped by game. They could be within different CP0, CP1, DD1, DD2 parent still, if we make sure people are not allowed to send PRs with changes in the MRA-Alternatives ZIP file only.

But if we want to cover all angles, the folder structure would be an exact 1:1 copy between the the mra/_alternatives folder and MRA-Alternatives_XXXXXXXX.zip/_alternatives, so in the case of somebody sending a PR with a new version of that zip, but no changes in the mra folder, that could also be covered and synced via bot.

gyurco, 2020-06-30:
Storing zipped MRAs are really defeating the purpose of git, you cannot see the change history in it. I would rather fix the shortcomings of the updater script than abusing git.

theypsilon, 2020-06-30:
I don't necessarily agree with having ZIP files for _alternatives, but that's the way it is inherited, and it works fine for distribution (it's faster). I agree it's smarter having files that are diffed by git and have a proper git log, but still ZIP generation could be automated thanks to Github Actions without human intervention. The bot could perform one-way sync or two-ways sync. So, having the ZIP thing is not that important in the end of the day, because automation.

It is more relevant the internal structure of the _alternatives folder, which is what I intended to discuss more in the previous post. Do you see any drawbacks in having the alts grouped by game like I've shown in my previous example @gyurco ?

gyurco, 2020-06-30:
Personally I have no objections against the per-game _alternatives.

theypsilon, 2020-06-30:
Ok, let's see what @jotego thinks about it. I'm up for doing all required changes next week.

theypsilon, 2020-07-18:
I've done the move of MRAs to per-game _alternatives here: https://github.com/jotego/jtbin/pull/114

jotego, 2020-07-22:
Let me first say thank you for taking the time to look at this. Let me write here my understanding of the repository:

-Original MRA files used during development are stored in the source code repository
-jtbin helps as a central download point, with not so much regard for history
-I would actually make it an HTTP/FTP server if it wasn't because MiSTer update script expects a github repository with certain folders
-I think we need to provide an independent MRA folder for MiST users but it could be built with symbolic links (maybe)
-The automatic zipping task helps MiSTer users

I'm not sure if there is any action pending on my side. Maybe trying the symbolic link approach?

theypsilon, 2020-07-24:
I just added another PR with MRA sync task here: https://github.com/jotego/jtbin/pull/117

After reading your message, I think there are still a few things to improve in MRA handling for your repositories. I wasn't aware you also have MRAs in each code repo.

At least, after my PRs, you will only have one duplicated in jtbin instead of two or more different ones, as there was before. And since it seems that jtbin is receiving PRs with MRA changes from the people, this will help with maintenance hopefully.

theypsilon, 2020-07-28:
I'm clossing this issue, as the duplicates should be correctly handled right now with these automatic tasks.

closed 2020-07-28 by theypsilon