jotego/jtbin #119: Some cores not loading

manast, 2020-07-29:
Probably this is something specific to my setup but I haven't seen this behaviour before. I am using the latest version of everything, mist firmware, cores and arc files. What happens is that games such as Black Tiger, Double dragon, or Tiger Road do not even try to load when I choose the core in the OSD, the OSD just closes and nothing happens. However games such as Gng, Bubble Bobble loads without problems, likewise the CPS1 games that also load as well.

What can cause this issue?

jotego, 2020-07-29:
ARC files with long lines for DIP switches cause problems in the firmware. Using the latest version of the MRA tool should fix this problem. I have just zipped the ARC files I have in the root of my SD card so you can compare. 
I am thinking of distributing MiST binaries with the ARC files to avoid this problem but as the MRA tool becomes more stable it may not be needed.
Let me know how it goes.
[mist-arc.zip](https://github.com/jotego/jtbin/files/4994807/mist-arc.zip)

manast, 2020-07-29:
I tried with the latest version or mra tools as well as with your arc files and the result is exactly the same, nothing happens and after 3-4 seconds the MiST is reset...

jotego, 2020-07-29:
Maybe your downloaded RBF files are wrong. If you download directly from the browser you may get the HTML instead of the binary file sometimes.

manast, 2020-08-08:
Closing this, the error was that the .rbf file I used did not match the one in the .arc file.

closed 2020-08-08 by manast