jotego/jtbin #120: Bionic Commando - Missing directory: bionicc

FiZiX, 2020-08-05:
Trying to play Bionic Commando on MiSTer. It says the bionicc directory is missing despite the fact that I have bionicc.zip in games/mame/ and it appears to match the MRA.

jotego, 2020-09-09:
I don't include md5 numbers in MRA files because MiST and MiSTer process the md5 in a different way. CRC values for indivudal files are included.
Please make sure to use a merged zip file (bionicc.zip). Also check that there are no other "mame" folders else where. MiSTer can be a bit picky when looking for the mame files. 
I don't think there is any problem with the RBF/MRA files themselves so I'm closing the issue.

closed 2020-09-09 by jotego

FiZiX, 2020-09-10:
@jotego Sorry but I'm afraid I must be misunderstanding something. I see md5 hashes in all of your MRA files.

Ex: [https://github.com/jotego/jtbin/blob/master/mra/CPS0/Bionic%20Commando.mra#L11](url)

Also, I'm using the mame-getter scripts to download the merged zip files, which seems to be working for all of your other cores. I'm not sure why I'm having an issue with this core in particular.