jotego/jtbin #121: Combat School and Contra no image.

Akaifutari, 2020-08-07:
Hi José.

I have the same problem with this two games. When I load the cores in my cabinet it only appear a black screen all the time. 
It's weird because nothing appear on screen, not even the menu options of the MiSTer, I only can reset the FPGA with the phisical button.
I revised the mra, rbf and zip folders and everything is in the correct place.
I also tried zip files from differents sets, but the problem is the same.

jotego, 2020-08-07:
If you cannot get the mister menu it means that the RBF file was not found. If it is in the right place, it might be corrupted. 
If you can get the menu and see the game title on it but the game doesn’t boot, it means that the zip file with the ROM was not found.

Akaifutari, 2020-08-08:
Hi again.

I downloaded the files from your links, also a friend of mine sent me his files and roms and doesn't work neither.
Now I have created a new Micro SD  from the beginning only with Combat School and Contra in the SD card and the problem still there.

jotego, 2020-08-08:
Can you get MiSTer menu to show up now at least? Or is it just a blank screen?
You can get information about why MiSTer is failing by looking at the console following these instructions: https://github.com/MiSTer-devel/Main_MiSTer/wiki/Console-connection

In my particular setup I have the MRA in _JT and the RBF in _JT/cores. ROM files are in games/mame. 

My MiSTer.ini is very simple too:

[MiSTer.ini.zip](https://github.com/jotego/jtbin/files/5045235/MiSTer.ini.zip)

Akaifutari, 2020-08-08:
I can't see the MiSTer's menu now, I have the same problem than before.

I was think that maybe this is a resolution or refresh problem, it's only occurs wiht Combat School and Contra. 
Those games uses the same resolution or refresh? Anyway is weird because I'm using an arcade cabinet with his original arcade monitor (Nanao MS9-29 bi-frequence 15 Khz / 24 Khz)

Akaifutari, 2020-08-08:
I've discovered something new. I've configured the buttons on both cores and I can hear everything (coin sound, music, effects..) but I can't see nothing.

jotego, 2020-08-09:
Try getting into the OSD menu -even if you get no image- and go to the second line, where V-sync adjust is. Try moving blindly throught it. Take into account that it will take 4-5 seconds for the TV to adjust to the new setting. Hopefully one of those settings will work with your screen.
Contra uses overscan, but Combat School doesn't. So I would assume the later should work flawlessly if this was a screen-core adjustment problem.

Akaifutari, 2020-08-09:
Hi Jotego.

Problem solved!!

The problem was the interface that I use for connect the MiSTer to my cabinet. Normaly I use a JammASD and for whatever reason that interface is not compatible with some resolutions. 

Luckyly I have another interface, a JammaCON. I've connected the MiSTer with that interface and the problem desappear. Now I can see the cores working flawlessly in my cabinet.

Thanks for your help!

closed 2020-09-09 by jotego