jotego/jtbin #122: Trojan - Elevators play double sound effect

Aquas77, 2020-08-10:
I thought this was odd and something I hadn't heard before. The elevators seem to double up their sound as you ride down them. 

As heard in this clip using Trojan (US set 2):
https://clips.twitch.tv/LittleTrustworthyStingrayCorgiDerp

Still on 32mb ram

jotego, 2020-09-09:
I think that is genuine to the original game. I'll keep the issue open in case we find more evidence about this.

jotego, 2021-01-16:
Could you try with 20210115 version (yesterday's)? It could have been related to the YM2151 timer. It seems like the clip link is broken now. I guess those are not permanent.

Aquas77, 2021-01-31:
Hello, apologies for the late response here.  I tried it and the issue is still present unfortunately. The clip link should work, by the way. Maybe it wants you logged in?