jotego/jtbin #123: Trojan - graphic glitch - white lines on bottom left of screen

Aquas77, 2020-08-10:
Horizontal small, white lines sometimes flicker in the bottom left of the screen at the edge, which I had not seen before in the Capcom Arcade Cabinet port on PS3. I can find another occurrence of it if needed, but I have one clip for now. Seems to occur as action is going on or scrolling.

https://clips.twitch.tv/PolishedEndearingBeaverNinjaGrumpy

jotego, 2020-08-14:
Thank you. I actually hadn't noticed them and wasn't aware. I have made some improvement to the sprite generator for Side Arms, making it more robust. It is the same circuit used for Trojan too so I'll update the core soon and see if that helped.

jotego, 2020-09-09:
I think this has been corrected with the improvements made lately to the CPS0 sprite generator. It seems ok now. https://github.com/jotego/jtbin/commit/e2f3272c5fcccdd0e116ff2a58635fea39bf1257

closed 2020-09-09 by jotego