jotego/jtbin #125: [MIST] CPS1 instability

squidrpi, 2020-08-13:
CPS1 core on the MiST is quite unstable, it regularly freezes. The MiST itself keeps working fine, i.e. F12 menu.

Noticed the instability occurring when SF2 support was added, before then it was rock solid.

jotego, 2020-08-13:
I’m a bit skeptic about that. The MiST compilation passes all timing constraints so I would rather suspect of something else. 
Note that games often have a freeze DIP switch that you Mary set inadvertently. You may hit pause too and pause the game, although I think the MiST version shows the credits screen in that case. 
You may also have a corrupted RBF file or ROM set. SD cards are not very reliable. Finally, bootleg versions may not work correctly on the core as only the original hardware is implemented.

jotego, 2020-08-15:
I have updated the file using a suggestion from Gyurco (slingshot) which may help. The old files passed timing analysis, the new ones too but, well, have a go at them.
https://github.com/jotego/jtbin/tree/master/mist/cps1/releases

squidrpi, 2020-08-16:
No difference for me. Can you attach the pre-SF2 built core? I remember that one being a solid build. Can't seem to access the old builds anymore.

jotego, 2020-08-16:
You can find the old builds if you browse the old posts in Patreon. I don't have them handy right now.

I'm still a bit skeptic about this being a problem in the core. You may want to try with a different SD card, etc.