jotego/jtbin #135: Bionic Commando graphic issue

Aquas77, 2020-10-18:
In Stage 4 the big moving grey doors have some graphic problem / flickering going on the tops of them, as seen in this video a few times. I don't believe this should be normal... never seen that before.

https://youtu.be/dc--BcccXmM?t=350

Still using 32mb MB ram, and pretty sure I'm on the latest RBF.

jotego, 2021-01-11:
I thought this was a latency problem with the SDRAM: tile data arriving too late. I changed the core to work with the bank-interleaving controller (commit https://github.com/jotego/jt_gng/commit/fa46a8b5aa8a089dd885e1aeea22eea856f7888b), which speeds up latency 4x. However, it is still the same. So the cause lies somewhere else. I will continue looking into it.

jotego, 2021-12-22:
I think this has been fixed for a while. It is ok in version e1921500 and probably was fixed much before that.

closed 2021-12-22 by jotego