jotego/jtbin #138: CPS1: graphical glitches with Street Fighter 2 Mix

mrfpga, 2020-12-05:
Street Fighter 2 Mix is a fan created hack based on Street Fighter 2 Champion Edition for the CPS1. The hacked romset sf2mix.zip can be obtained from https://sf2mix.github.io/. In order to play the game the hack author recommends to replace sf2mix.zip with sf2ce.zip. That way the games works with Jotego's core on MiSTer. However, there are some graphical glitches in the backgrounds of the stages.

Chocobobobo, 2020-12-11:
Jotego already mentioned many times that he won't support hacks. 1.5 bootlegs barely got a pass a few months ago... He would better spend his time on CPS2 or PGM, than supporting the 56432115 SF2 hacks around. Just my two cents as a loyal patreon.

jotego, 2020-12-12:
As @Chocobobobo says, I have given up on supporting unofficial game versions that require hardware modifications. Particularly, bootlegs. 

However, if someone makes the changes to the core for me and those integrate well with the original hardware, then I will take those changes in.

I will leave this issue open for a while in case someone accepts the challenge.

mrfpga, 2020-12-15:
Thank you jotego for your reply. Meanwhile the author of the hack has released a new version where the fake 3d backgrounds which caused graphical glitches on your core can be turned off. Without these fake 3d backgrounds, the hack is perfectly playable on your core. I have therefore cretaed a pull request (#146) to add the hack's mra to the /mra/_alternatives/ folder.

As for hardware modifications, I think the hack doesn't require any but runs on stock hardware. I don't know why the fake 3d backgrounds aren't working. It could theoretically be an inaccuracy of your core which just isn't visible with other games. May it be as it is, it is your time and I respect your decision not to look into this any further. I am very grateful for your work and looking forward to your next releases.

jotego, 2020-12-16:
Thank you for your PR. I have already accepted it. I didn't know the game ran on a regular CPS1 board. In that case, I am more interested in it.
If some graphics should show and are not, it might be a layer priority problem. That is actually configured in the MRA file. The bytes that say "CPS-B config for ..." follow basically the same format found in the file video/cps1.cpp of MAME. They contain the address of each CPS-B register for the game and the location of some bits.
I think it might be the case that that information is not right in your MRA file.
Another possibility is a priority bug in the core. I know there is at least one such problem (see https://github.com/jotego/jtcps1/issues/40). But, even that one may be solvable by modifying the MRA file.

closed 2021-03-05 by jotego