jotego/jtbin #140: CPS 1.5: crashes in Punisher and Cadillacs and Dinosaurs

rarcos, 2020-12-11:
These crashes appear in release 201211.

There are random crashes in gameplay in both games:

[image]
[image]
[image]

Also, the games crash always at the same part in attract mode:

[image]
[image]

Also, Cadillacs and Dinosaurs shows this behaviour: enter the first building, defeat all enemies, but don't get close to the window and the the time run out, when the player respawns the sprite will become unstable and then disappear, a few seconds later, the next scene will start. I'll try to capture footage from this if I can reproduce it again.

rarcos, 2020-12-11:
Every crash happens with a 128 MB memory module.

rarcos, 2020-12-12:
Unable to reproduce with 32 MB board.

jotego, 2020-12-12:
I honestly think this is an electrical problem in some modules. It seems to be related to the data pattern across the bus. In the past, I have got away with it by holding the bus to a fixed value in the SDRAM idle cycles. But after increasing the access performance there are fewer of these cycles. Holding the data bus doesn't seem to help anymore.

I could arguably try to reduce the data speed as there is margin for that in the CPS 1.5 core. But I would just be pushing the problem forward. CPS2 will need more data speed and so will other future cores.

I am running some more tests today but I do not have much hope in being able to solve this from the FPGA side. I do think this is a module problem.

jotego, 2020-12-12:
@rarcos, one question. If you upload the game and watch the demo running: do you see the errors occur always at the same location?

jotego, 2020-12-12:
@RetroShopPT, could you gather PCB parasitics from the layout tool, please?

RetroShopPT, 2020-12-12:
Will work on it @jotego

VirtuaVespa, 2020-12-12:
Punisher, W.o.F, Muscle Bomber and C&D appears to be fine. I couldn't reproduce the Cadillacs time out bug. I only have an issue with Slam Masters. Select a single match, lose the match, then continue. Character select screen will display something like "ILLEGAL INST. 0000F92A" or "ADDRESS ERROR 00010A2B"

SegaSnatcher, 2020-12-13:
I'm having an issue with Saturday Night Slam Masters where the game speeds up.  Its hard to explain, but the animations and such speeds up vs what I'm seeing via youtube Mame emulator runs.

jotego, 2020-12-13:
Could you try updating? Manually remove the old cores first if possible just to be sure. 
Technical details for developers:
The SDRAM clock is now generated by a DDR cell in the pin. This gives an uniform SDRAM clock delay at the pin with respect to the main clock. But data is sampled now at the SDRAM negative clock edge. This means that the last word of the burst is being read with the bus in high impedance (!)

rarcos, 2020-12-13:
After updating to latest release, Punisher still freezes in the same location in demo mode, with same message (ILLEGAL INST. 0005CFA2). Cadillacs and Dinosaurs also always freezes in the same place in demo mode.

Before both of those games trigger freezes in demo mode, I'm able to play them and even have finished both of them.

Muscle Bomber Duo now can boot some times instead of freezing at the beggining. Slam Masters doesn't have sound issues anymore at title screen.

An interesting find: when a game crashes or freezes, succesive loading of other games also always crash at some point in gameplay or demo mode, until I turn off my MiSTer for a few minutes.

jotego, 2020-12-13:
Reported by @rarcos via Discord:

The Punisher doesn't always freeze, but when it does (during the demo) is at a consistent location:
[image]

jotego, 2020-12-18:
Please update to today's release and answer this poll:
https://www.patreon.com/posts/45183707

movisman, 2020-12-19:
Hi there,

Thanks for all your incredible work on these arcade cores. I've only just got into MiSTer and they are so good.

Just to echo some of the above findings with my XS-D v2.5 128MB Module.

If the MiSTer is booted 'from cold', Cadillacs and Dinosaurs will make it to attract mode. However, if it has been switched on for a while, Cadillacs and Dinosaurs will always fail in the same spot. It happens just as the title screen transitions to attract mode. The screen goes blank, and "illegal inst. 0009A90C" is printed. However, if I don't let it get that far - eg. I insert a coin and start the game, I can actually play through to the end without issue. This is what I see:

[image]

Regarding the new SF Zero CPS 1.5 hack, I cannot test this, as it crashes every time 20-30 seconds after the ROM loads. The following screen pops up, and the digits in column +8, row 00FF8010 just keep whirring around various values in a continuous, fast loop:

[image]

The Punisher and Warriors of Fate seem ok for me so far, no illegal instructions that i've seen yet in attract mode - but I haven't tried to actually play these through to the end yet.

Muscle Bomber Duo and Slam Masters seem to have their own issues which I know you have mentioned already, but I do get memory related errors on these cores at random points, either in attract mode or in game. These aren't playable at the moment, they are guaranteed to crash with illegal instruction messages.

To confirm I am running the latest (as of 19th Dec 2020) 20201218 version.

I assume my module is probably unable to sustain the 96MHz speed you mentioned on your Patreon update, causing these problems. I may look to get it swapped for another (or possibly even a 32mb module), as this module is only a few weeks old.

Regarding @rarcos and @SegaSnatcher 's comments above:

> An interesting find: when a game crashes or freezes, succesive loading of other games also always crash at some point in gameplay or demo mode, until I turn off my MiSTer for a few minutes.

I find this exact behaviour as well.

> I'm having an issue with Saturday Night Slam Masters where the game speeds up. Its hard to explain, but the animations and such speeds up vs what I'm seeing via youtube Mame emulator runs.

I have also noticed this on Muscle Bomber Duo, everything goes hyperspeed and the match ends in a no contest. It is most odd.

---

Hope some of this information is useful. I'm sure it is all memory related and my module just cannot keep up.

If you need any further info or testing, just let me know.

Thanks a lot