jotego/jtbin #144: Saturday Night Slam Masters hangs up afterfight

jotego, 2020-12-13:
An illegal instruction message shows. This can be replicated with 32MB/128MB modules and on MiST so I do not think it is related to the SDRAM.

[image]

SegaSnatcher, 2021-01-17:
Using latest CPS 1.5 96Mhz release I do not get this error anymore.  I was able to play through many matches with no issues.

odiaboeeu, 2021-01-18:
Yes, Yesterday I managed to finish without any error in this latest version of the core that Jotego released 😇🙏

jotego, 2021-01-19:
I think this has got fixed accidentally, or maybe this was a manifestation of an SDRAM problem that is less sensitive in this release. I need to check the changelogs to make sense of this.

closed 2021-01-31 by jotego