jotego/jtbin #145: V-Sync issues with Cadillacs and Dinosaurs and Warriors of Fate

Alex-Catalin, 2020-12-14:
As the title suggests, I noticed some V-Sync issues with these 2 games. They appear always at the same time, in the same places in the game.

For example, in Cadillacs and Dinosaurs, they can be seen right at the beginning, in stage 1 and they appear again in the second level, when you are in the swamp. On Warriors of Fate, they look even worse than on Cadillacs and Dinosaurs.

I have not experienced any other issues, such as crashes or anything else, just these V-Sync problems which are consistent and appear every time in the exact same places.

paulb-nl, 2020-12-19:
I also noticed this with Cadillacs and Dinosaurs. The tearing does seem to go away if Turbo is enabled. Here is an example:
[image]

jotego, 2020-12-20:
Thank you very much for that screenshot. I actually needed it. I think this comes down to DMA accuracy. I may delay working on the DMA until I get the first CPS2 beta out. This affects all CPS1/1.5/2 titles so stay assured I will get it corrected.

Alex-Catalin, 2020-12-21:
@paulb-nl do you get this with the core that added support for QSound on Street Fighter Zero as well, the one from 18th of December? I am asking because yesterday I played a bit and I couldn't see the screen tearing anymore on Cadillacs and Dinosaurs and I could always see it with the first version of the core, always in the same places.

But if this is a DMA accuracy this, it also means that there is a chance to happen randomly...

paulb-nl, 2020-12-22:
Yeah the screenshot is from the core of 18th of december.

jotego, 2020-12-30:
It's quite odd, indeed. CPS can set the scroll of each line independently for one of the tilemaps, and that information is read at the beginning of the line via DMA. Somethings seems to be going wrong about it there. I will be reviewing the DMA circuit today.

jotego, 2021-01-31:
This is fixed in today's release.

jotego, 2021-02-07:
A similar issue can still be seen on occasion. Check [video](https://t.co/MFMeV2A9A3?amp=1).

This time it doesn't seem to be related to the CPU but to the DMA missing data, probably because of SDRAM congestion.