jotego/jtbin #148: 3rd player controller not recognized.

SumolX, 2020-12-19:
In the main menu I can assign the inputs and reassign per game.  However, in test mode and in game none of the buttons are being recognized.

jotego, 2020-12-19:
Excuse me, but I cannot which game are you referring to?

SumolX, 2020-12-19:
I tried C&D and warriors of fate.

Epyc7, 2021-01-12:
I can confirm this also.  Configured Player 3/4 controls but cannot insert coin for Player 3/4 in Cadillacs and Dinosaurs. I assume this is the same for other 3/4 player games as well.

However, Final Fight Anniversary Edition 3 Player works as expected.

jotego, 2021-01-31:
Update: the memory map seems ok. Button mapping too. The culprit is still unclear.

uncon, 2021-02-01:
I can confirm the same behavior. I have been unsuccessful in getting >2 simultaneous player inputs working for CPS games.

jotego, 2021-02-02:
I think it is a specific problem of CPS1.5. The 3-player games in CPS1 should be ok.

uncon, 2021-02-02:
I just confirmed that, indeed, CPS 1 seems alright, and only the CPS 1.5 games I've tested exhibit this issue.

jotego, 2021-03-05:
Technical notes:
The CPU does access the 3rd player inputs in simulation. I can see joy3_cs going high.

manast, 2021-04-17:
In Cadillac & Dinosaurs I noticed that you can insert coins for player 3 with the "9" key instead of "7" that would have been the correct mapping. Unfortunatelly I could not find the key for start so I could not test the other buttons. Maybe it is just a matter of wrong mapping?

manast, 2021-04-17:
forget my previous comment, "9" seems to mean "insert coin for all players"... 
But I can confirm that on Knights of the round all three players work perfectly.

jotego, 2021-05-20:
Fixed in https://github.com/jotego/jtcps1/commit/63db0d01d1ce665856c5e88e9b2e7feec5a214d5
Thanks to blackwine for the edit.
I will update the binaries shortly.

closed 2021-05-20 by jotego

trukjr, 2021-05-25:
C&D still does not work with 3 players after the Update_All script was run for the MiSTer.

jotego, 2021-05-27:
Please try updating again, maybe something went wrong. I have checked it here and I can control the third player correctly. 
Could this be related to your setup?

trukjr, 2021-05-28:
> Please try updating again, maybe something went wrong. I have checked it here and I can control the third player correctly.
> Could this be related to your setup?

Hey Jotego,

I ran the Update_All script last night and both the CPS1 and CPS15 files updated correctly during the Jotego section.  Afterwards,  C&D worked perfectly with 3 players!  I'm not sure if there were any changes made since I ran this on Monday.  Thanks!